MrYsLab/pymata-express #1: Auto-discovery issue on macOS Mojave (Version 10.14.2)

ilovetogetspamed, 2019-02-05:
I installed pymata-express on my iMac in a virtual env.  I installed the Arduino firmware onto an Arduino Mega2560 (R3) and tried running an example.

<code>$ python retrieve_capability_report.py</code>

This didn't work as expected.

The documentation states:

> By accepting the defaults, PymataExpress will attempt its auto-discovery procedures to locate and connect to an Arduino. This assumes that you have the FirmataExpress sketch uploaded to the Arduino.

when I run the example program I get:

```
$ python pymata-express/examples/retrieve_capability_report.py 

Pymata Express Version 1.00
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening all potential serial ports...
	/dev/cu.usbserial-AH06VXD3
	/dev/cu.usbmodem146701

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1

```
and it just hangs there so I have to kill the program with Ctrl+C.

I added the com_port parameter to the initialization and it worked just fine:

```
# instantiate pymata_express
board = PymataExpress(com_port='/dev/cu.usbmodem146701')
```
Something in the default initialization is not working on macOS.

Thanks for all the work on the pymata libraries.

Let me know if you need me to test anything.

ilovetogetspamed, 2019-02-06:
> @ilovetogetspamed <https://github.com/ilovetogetspamed> Thanks for letting me know. Could you try:
>
> board = PymataExpress(arduino_wait=10)
> and let me know if it still hangs. Tnx.

Alan,

I got the same result.  Nothing happens.

Chris

MrYsLab, 2019-02-06:
Thanks. I am going to have to analyze the code - it should never hang. I retested on my 2560 to make sure that I did not miss anything when I tested, and it works fine. That is on Ubuntu Linux, so I will retest on Windows and also on a Raspberry Pi to see if I can recreate the problem. One more favor to ask - could you try running the [blink example](https://github.com/MrYsLab/pymata-express/blob/master/examples/digital_output.py). Feel free to change the pin number - I tried with pin 13. If this works (I don't think it will), please provide python script that causes the hang.

ilovetogetspamed, 2019-02-06:
Alan,

i made some progress...  I think the trouble is with multiple devices.

$ python blink.py 

Pymata Express Version 1.00
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening all potential serial ports...
	/dev/cu.usbserial-AH06VXD3
	/dev/cu.usbmodem146701

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1
Traceback (most recent call last):
  File "blink.py", line 49, in <module>
    board = PymataExpress()
  File "/private/tmp/pymata-express/venv/lib/python3.7/site-packages/pymata_express/pymata_express.py", line 176, in __init__
    self.loop.run_until_complete(self._find_arduino())
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 568, in run_until_complete
    return future.result()
  File "/private/tmp/pymata-express/venv/lib/python3.7/site-packages/pymata_express/pymata_express.py", line 355, in _find_arduino
    i_am_here = await self.serial_port.read_until(expected=b'\xf7')
  File "/private/tmp/pymata-express/venv/lib/python3.7/site-packages/pymata_express/pymata_express_serial.py", line 160, in read_until
    if not self.my_serial.in_waiting:
  File "/private/tmp/pymata-express/venv/lib/python3.7/site-packages/serial/serialposix.py", line 467, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
OSError: [Errno 6] Device not configured

(above) WHEN I PULLED THE PLUG ON /dev/cu.usbserial-AH06VXD3

then

$ python blink.py 

Pymata Express Version 1.00
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening all potential serial ports...
	/dev/cu.usbmodem146701

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1

Arduino found and connected to /dev/cu.usbmodem146701

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 FirmataExpress.ino
Auto-discovery complete. Found 70 Digital Pins and 16 Analog Pins

ON
OFF
ON
OFF
ON
OFF
ON
OFF

WHEN ONLY THE Arduino Mega2560 was plugged in.

Hope this helps.

Chris

MrYsLab, 2019-02-06:
Great. That gives me a place to start looking.  Is  /dev/cu.usbserial-AH06VXD3 also connected to an Arduino? If not, what type of device is connected?

ilovetogetspamed, 2019-02-06:
It is a SparkFun FTDI Basic (https://www.sparkfun.com/products/9873 <https://www.sparkfun.com/products/9873>) plugged into a SparkFun Pi Wedge (https://www.sparkfun.com/products/13717 <https://www.sparkfun.com/products/13717>) which is connected to a Raspberry Pi 3 Model B Rev 1.2

MrYsLab, 2019-02-06:
Thanks. I am able to recreate the problem by plugging in an ESP8266 as my first USB/Serial device and then the Arduino. Now I just have to figure out why the built-in timeout in get_firmware_verison() is not firing.

ilovetogetspamed, 2019-02-06:
Alan,

I did some further testing, I plugged in three Arduinos into a hub.  The original using instance ID 1 and one using instance ID 2, and left the third with whatever code it had on it (not FirmataExpress)

I got:

$ python blink.py 

Pymata Express Version 1.00
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening all potential serial ports...
	/dev/cu.usbmodem1466201
	/dev/cu.usbmodem1466301
	/dev/cu.usbmodem1466101

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1

Arduino found and connected to /dev/cu.usbmodem1466101

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 FirmataExpress.ino
Auto-discovery complete. Found 70 Digital Pins and 16 Analog Pins

ON
OFF
ON
OFF
ON
OFF
ON
OFF

Also, the version using arduino_wait worked too.

So it looks like the code is getting confused by non Arduino signatures perhaps?

Have fun debugging.

Chris

ilovetogetspamed, 2019-02-06:
You're welcome!

MrYsLab, 2019-02-06:
I figured out what is happening. When I query for the Arduino instance ID using the read_until method in the express serial module, nothing comes back and I wait forever. So now, I just need to provide a graceful way of timing out. I have 2 solutions in mind. I will get back to you when I hopefully have a reliable fix.

MrYsLab, 2019-02-06:
Version 1.2 is now on pypi and this should fix the issue. Please update your distro for pymata-express and this should fix the problem. If the problem is fixed for you, please let me know.

Thanks again for letting me know about the issue.

MrYsLab, 2019-02-07:
I am going to close the issue. If you are still having problems, please post a reply here and I will see it even though the issue is closed. I can then reopen the issue if your problem was not solved.

closed 2019-02-07 by MrYsLab