MrYsLab/pymata-express #5: firmata express on arduino due :)?

willmendil, 2019-07-23:
Hi again, 

Whilst waiting for firmata, I've been playing around with firmata express on the mega, and I'm quite impressed. Wondered, if adapting it for the due would be doable as well as firmataplus? 

I wonder whether the faster clock on the atmel would make firmataexpress even "expresser".

MrYsLab, 2019-07-23:
This should not be a problem, but first, the fix from the Firmata folks needs to be made available. Just need to remove the tone library import and it should be good to go.

willmendil, 2019-07-24:
wanna give it a try for express as well?

MrYsLab, 2019-07-24:
If you specify the com port and baudrate (57600) as explained [here](https://mryslab.github.io/pymata-express/guidelines/#parameters-com_port-and-baud_rate) I believe it should work.

willmendil, 2019-07-24:
No, when I upload firmataExpress to the arduino Due, I get: 

```
Arduino: 1.8.9 (Windows 10), Board: "Arduino Due (Programming Port)"

FirmataExpress:40:21: error: avr/wdt.h: No such file or directory

 #include <avr/wdt.h>

                     ^

compilation terminated.

exit status 1
avr/wdt.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

you had answered 

> The cause of the error you are seeing with FirmataExpress is because the Due does not have a watchdog timer feature and so the compiler cannot find an equivalent function to support the Due.

Is there a work around to get express on the Due?

MrYsLab, 2019-07-24:
I was not very clear. You should be able to use FirmataPlusDue from pymata-aio instead of Firmata Express. You just need to explicitly use the com port and speed parameters in your Python script using pymata express. I added those options so that StandardFirmata (assuming that it compiles on all platforms) could be used. Since FirmataPlusDue is using an older version of StandardFirmata as its basis, this should work. I have tested using StandardFirmata on an Uno and it did work for the StandardFirmata feature set.
Let me know if this is still not very clear and I will further explain.

willmendil, 2019-07-24:
oh, right, well I uploaded your new and gorgeous FirmataPlusDue on the Due. but it seams there's an issue regarding the arduino ID you use in pymata express

```
Pymata Express Version 1.3
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening /dev/ttyACM0 ...
Waiting 4 seconds for the Arduino To Reset.
Traceback (most recent call last):
  File "mve.py", line 107, in <module>
    my_board = PymataExpress(com_port="/dev/ttyACM0", baud_rate=115200)
  File "/home/pi/robopi/envrobopi/lib/python3.7/site-packages/pymata_express/pymata_express.py", line 185, in __init__
    self.loop.run_until_complete(self._manual_open())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/home/pi/robopi/envrobopi/lib/python3.7/site-packages/pymata_express/pymata_express.py", line 408, in _manual_open
    raise RuntimeError('Invalid Arduino ID reply length')
RuntimeError: Invalid Arduino ID reply length

```

MrYsLab, 2019-07-24:
Try changing  my_board = PymataExpress(com_port="/dev/ttyACM0", baud_rate=115200)
to
 my_board = PymataExpress(com_port="/dev/ttyACM0", baud_rate=57600)

willmendil, 2019-07-24:
SUCCESS, my bad I did't realise the difference in baud rate between the two. 
now, I only have the issue on stackoverflow :)

closed 2019-07-24 by willmendil

MrYsLab, 2019-07-24:
Did my answer on StackOverflow help? I only use zmq in a pub/sub configuration, so I am not very familiar with req/rep.

I have a [python-banyan gateway](https://github.com/MrYsLab/python_banyan/blob/master/projects/OneGPIO/arduino_uno/arduino_gateway.py) that uses asyncio to talk to pymata-express.

It may seem a  little convoluted because I use 2 levels of inheritance, but is actually fairly straightforward. This implements a GPIO abstraction protocol I developed called [OneGPIO.](https://mryslab.github.io/python_banyan/#gpio_intro/)

willmendil, 2019-07-24:
I did try your answer, but got the same issue. I've uploaded a working example if you want to try it out. for the REQ side, you just need to loop over the send and recv with a 1second time.sleep ( i can upload that too if youneed). I'll take a deeper at the gateway, but at first glance it didn't help

MrYsLab, 2019-07-24:
I removed the references to wdt and tone in firmata-express and placed it in https://github.com/MrYsLab/FirmataExpressDue

It compiles. I am guessing that something changed in the StandardFirmata files since I created firmata-express. 

If you do download from the green button on github, just rename the directory from FirmataDueExpress-master to FirmataDueExpress after unzipping.

If you can provide code that I do not have to modify I can take a quick look.

 I don't know if you are using an IDE for python development, but if not, you may wish to look at [PyCharm. ](https://www.jetbrains.com/pycharm/) The debugger is excellent. There is a free version called the community version. The professional version is available for free for University students/staff. The main difference between the two is that the professional version allows you to debug on a remote computer like the Raspberry Pi from your PC. Unfortunately PyCharm is too heavyweight to run on the RPi directly. I use the remote debugging feature and could not live without it.

MrYsLab, 2019-07-24:
Forgot to mention, you did not need to use any command line parameters with FirmataExpressDue. It should run full speed and the correct serial port automatically.

willmendil, 2019-07-25:
FirmataDueExpress works as well with ping and sonar. We a re all good :)
I just installed pycharm, thourgh uni i have the student version for 1 year just need to know how to use it. I had tried it but it was heavy and bulky compared to atom. I'll give it another try. 

I'll put together the a script you can try regarding the issue I have on stackoverflow. Where do you want me to put it?

MrYsLab, 2019-07-25:
If you are happy with Atom, there is no reason to switch to PyCharm. I am glad that the due sketch works. I will do some additional testing using a similar release for the Uno and if all looks good, I will make the sketch downloadable through the Arduino IDE.

I sent you some email and did not see that you put the files on stack overflow. I will take a look and let you know if I found the cause.

willmendil, 2019-07-25:
you sent me an email? where?

MrYsLab, 2019-07-25:
The email on your github accout: watkins@insp.upmc.fr
I hope that was ok.

 I am looking at your code. I need to find my hc-sro4 and hook it up to test.
Will get back to you later on this. Do you prefer an answer on StackOverflow?

willmendil, 2019-07-25:
oh, yeah, don't use that email: prefer will.mendil@gmail.com
And yes, lets do that on stackoverflow so the discussion is saved over there for others.