MrYsLab/pymata-express #6: [feature] reconnect to arduino if disconnected

willmendil, 2019-07-25:
Me again, this is just a feature that could be helpful. Basically, a loop that checks if the arduino is connect and reconnects if it has been disconnected.  I'll try to integrate this feature, but you know your code better than I do.

MrYsLab, 2019-07-26:
This might get messy on the pymata-express side. The Arduino's using Firmata have an issue, in that, once you enable an analog input pin, the Arduino will keep streaming the data. The only way to stop it is to do a hardware reset of the Arduino. There is a very high likelihood this streaming data will confuse pymata-express and it will raise an exception.

Some of the Arduino's, like the Uno, support a watchdog timer. If that expires, the board can be totally reset, but not all the Arduino's have this feature. 

Re-establishing connectivity should be possible, but resyncing the Arduino and pymata-express may be tricky. 

Give it a try - I might be wrong here.

MrYsLab, 2019-08-08:
Will, I would like to close this issue. I am not sure if you tried to accomplish this yet or not. If you are still planning on doing this, please make sure that your solution does not cause any breaking changes. If I close the issue, you can create a pull request at any time.

Is it OK to close this?
Thanks.

closed 2019-08-09 by MrYsLab