MrYsLab/pymata-express #8: hc-sr04_distance_sensor.py not returning values

MrYsLab, 2019-09-18:
Ping time needs to be adjusted to a longer delay.

willmendil, 2019-09-18:
What was the problem? Since you're saying, if I understand that the delay between the two triggers is extended. How many minimum pings can be done then? What's the minimum delay?

MrYsLab, 2019-09-19:
I was testing with a US100 sensor that supports 3-5v and did not receive any data back from the sensor. So I extended the timeout (I will be updating the API documentation). 

The delay is [tuneable](https://htmlpreview.github.io/?https://github.com/MrYsLab/pymata-express/blob/master/documentation/api/pymata_express.html#pymata_express.PymataExpress.set_pin_mode_sonar).

Is this causing an issue for you?

MrYsLab, 2019-09-20:
Will, I am closing this issue. Please let me know if you would like me to reopen it. Feel free to leave any comment here.

closed 2019-09-20 by MrYsLab