MrYsLab/pymata-express #9: how use stepper's  pins list

HotView, 2019-12-10:
I don't know how to connect to the motor, and what is the relationship between the pin parameter in the stepper function and the motor connection?
thanks！

MrYsLab, 2019-12-10:
Stepper motors require a driver board. This is [an example ](https://smile.amazon.com/KOOKYE-28BYJ-48-Stepper-ULN2003-Arduino/dp/B019TOJRC4/ref=sr_1_4?keywords=Stepper+Motor+28Byj-48&qid=1575978864&s=electronics&sr=1-4)of an inexpensive motor and drive electronics.

[This article](https://www.makerguides.com/28byj-48-stepper-motor-arduino-tutorial/) explains how to connect it to an Arduino.

HotView, 2019-12-10:
First of all, thank you for your detailed answers. I think I can learn more from you in the way of answering questions.

I know that to control the stepper motor, we need a pin to control the number of steps, a pin to control the direction, an enabled pin and GND. I don't know in this pins list, such as [8,9,10,11], which represents the direction, which represents the steps, which represents the enable, etc

HotView, 2019-12-10:
> Stepper motors require a driver board. This is [an example [...]

First of all, thank you for your detailed answers. I think I can learn more from you in the way of answering questions.

I know that to control the stepper motor, we need a pin to control the number of steps, a pin to control the direction, an enabled pin and GND. I don't know in this pins list, such as [8,9,10,11], which represents the direction, which represents the steps, which represents the enable, etc

MrYsLab, 2019-12-10:
The number of steps and direction is controlled by the software. Each of the 4 pins controls one of the motor coils and not direction. The [pymata-express API method for stepper_write](https://htmlpreview.github.io/?https://github.com/MrYsLab/pymata-express/blob/master/documentation/api/pymata_express.html#pymata_express.pymata_express.PymataExpress.stepper_write)  explains how to set the number of steps and the motor direction.

MrYsLab, 2019-12-10:
Here is [an example](https://github.com/MrYsLab/pymata-express/blob/master/examples/stepper.py) of how to use the stepper motor functions.

There are examples of pymata-express features [here](https://mryslab.github.io/pymata-express/examples/).

HotView, 2019-12-10:
@MrYsLab 

I read the blog about your recommendation just now. The connection method is Arduino + Stepper Motor Driver + Stepper Motor, as shown in the picture.
[image]
[image]
 
I don’t quite understand what you mean by “ Each of the 4 pins controls one of the motor coils and not direction”. 
You mean the arduino is directly connected to the stepper motor?

MrYsLab, 2019-12-10:
The stepper motor driver in firmata-express is based on the standard [Arduino stepper library.](https://www.arduino.cc/en/reference/stepper)

The article that I linked to in an earlier posting describes how to wire the ULN2003 motor driver board to the Arduino. I am not familiar with the driver board in your picture above, and I don't think it is supported by the Arduino stepper library, but uses a custom Arduino library. If you want to use that board, you will need to modify the firmata-express sketch to support it. This is a big job, but if you want to try and implement this, you can look at[ an article](https://www.instructables.com/id/Going-Beyond-StandardFirmata-Adding-New-Device-Sup/) I wrote several years back on how to add the Arduino stepper library to firmata as a guide.

HotView, 2019-12-10:
@MrYsLab 

Hello, I communicated with you on github just now, but since I am studying in China, I can't access websites like Google and github on my network, so the network is blocked again.

At present, I need to use Python to control an Arduino so that it can control the stepper motor to take different steps according to different external signals. However, after the discussion just now, I found that the stepper function does not support the motor driver I want to use.

I would like to ask whether the pins in firmata-express can be controlled to generate PWM?If this is possible, I can use a pin to generate a PWM wave to control my motor driver.

MrYsLab, 2019-12-10:
Yes, it supports PWM. Here is a [simple example](https://github.com/MrYsLab/pymata-express/blob/master/examples/analog_output.py).

MrYsLab, 2019-12-11:
I am closing this issue. If you have any additional questions you can post them here and I will see them.

closed 2019-12-11 by MrYsLab

Manish2208, 2020-03-28:
how to connect or control 2 or 3 stepper motor  using pymata ? Pls give some example for the same

MrYsLab, 2020-03-28:
@Manish2208 Hi again. None of my Firmata sketches support more than 1 stepper motor. If you are looking for a source about wiring up stepper motors, please check the following links:
https://learn.sparkfun.com/tutorials/easy-driver-hook-up-guide
https://learn.adafruit.com/adafruit-motor-selection-guide

Unless your project requires the precision of a stepper motor, you may wish to consider using either DC motors or Servo motors. Firmata can easily support a multiple of each type.