MrYsLab/pymata-express #15: Input callback is triggering multiple inputs.

teucrus, 2020-04-15:
First of all, I need to say how great this project is. Thank you for all your good work.
I would like to report an issue I am having.

Please consider the following code:

```
import asyncio
from pymata_express.pymata_express import PymataExpress

async def enableLed(data):
	print("-------------------------")
	print(data)
	print("-------------------------")

async def setup(board):
	await board.set_pin_mode_digital_input_pullup(40,enableLed)
	await board.set_pin_mode_digital_input_pullup(41,enableLed)
	await board.set_pin_mode_digital_input_pullup(42,enableLed)
	await board.set_pin_mode_digital_input_pullup(43,enableLed)
	await board.set_pin_mode_digital_input_pullup(44,enableLed)
	await board.set_pin_mode_digital_input_pullup(45,enableLed)
	await board.set_pin_mode_digital_input_pullup(46,enableLed)
	await board.set_pin_mode_digital_input_pullup(47,enableLed)

board = PymataExpress(com_port="COM9",arduino_wait=1)
loop=asyncio.get_event_loop()
loop.create_task(setup(board))
loop.run_forever()
```

This is the output of the callback when manipulating only one input (#41):

```
Pymata Express Version 1.11
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening COM9 ...
Waiting 1 seconds for the Arduino To Reset.

Arduino found and connected to COM9

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 FirmataExpress.ino
Auto-discovery complete. Found 70 Digital Pins and 16 Analog Pins

-------------------------
[0, 40, 1, 1586976240.831113]
-------------------------
-------------------------
[0, 40, 1, 1586976240.8779788]
-------------------------
-------------------------
[0, 41, 1, 1586976240.8779788]
-------------------------
-------------------------
[0, 40, 1, 1586976240.8779788]
-------------------------
-------------------------
[0, 41, 1, 1586976240.8779788]
-------------------------
-------------------------
[0, 40, 1, 1586976240.9319274]
-------------------------
-------------------------
[0, 41, 1, 1586976240.9319274]
-------------------------
-------------------------
[0, 42, 1, 1586976240.9319274]
-------------------------
-------------------------
[... 649 lines cut ...]
-------------------------
-------------------------
[0, 42, 1, 1586976275.9383821]
-------------------------
-------------------------
[0, 43, 1, 1586976275.9383821]
-------------------------
-------------------------
[0, 44, 1, 1586976275.9383821]
-------------------------
-------------------------
[0, 45, 1, 1586976275.9383821]
-------------------------
-------------------------
[0, 46, 1, 1586976275.9383821]
-------------------------
-------------------------
[0, 47, 1, 1586976275.9383821]
-------------------------

```

As seen in the above output although the pins are reported correctly (only pin #41 is changing), all of them trigger the callback. Also the pin is reported as a digital inpt (0) instead of digital input pullup (11).

Thank you again for your time and efforts.

MrYsLab, 2020-04-15:
Thanks reporting this. I am working on a fix and hope to have it available soon.

teucrus, 2020-04-15:
Thank you for the quick response. Because I need this working, do you have an idea where to look or a temporary workaround?

Thank you.

MrYsLab, 2020-04-15:
To get the pin type to report correctly, the following line needs to be added after [this line:](https://github.com/MrYsLab/pymata-express/blob/de8a710de8bfacf51cb983dfc53264b0a4b28699/pymata_express/pymata_express.py#L1238)
```
self.digital_pins[pin_number].pull_up = True
```

I am not sure about why the multiple reports, but it looks like I need to check for the previous value and if it didn't change, do not issue a callback. The code where this should be done is in [this method.](https://github.com/MrYsLab/pymata-express/blob/de8a710de8bfacf51cb983dfc53264b0a4b28699/pymata_express/pymata_express.py#L1475)

When the Arduino sends a digital pin change report, it sends it for a port that contains eight pins, so I need to look at this code in some detail to see how to affect the callback suppression. It should not be difficult, but I won't be able to look at this until later this evening or tomorrow.

MrYsLab, 2020-04-15:
You can try changing the async def _digital_message(self, data): method to this:
```
    async def _digital_message(self, data):
        """
        This is a private message handler method.
        It is a message handler for Digital Messages.

        :param data: digital message

        """
        port = data[0]
        # noinspection PyPep8,PyPep8
        port_data = (data[PrivateConstants.MSB] << 7) + \
                    data[PrivateConstants.LSB]
        pin = port * 8
        for pin in range(pin, min(pin + 8, len(self.digital_pins))):
            # get pin value
            value = port_data & 0x01

            last_value = self.digital_pins[pin].current_value

            # set the current value in the pin structure
            self.digital_pins[pin].current_value = value
            time_stamp = time.time()
            self.digital_pins[pin].event_time = time_stamp

            # append pin number, pin value, and pin type to return value and return as a list
            # if self.legacy_mode:
            #     message = [pin, value, PrivateConstants.INPUT, time_stamp]
            # else:
            if self.digital_pins[pin].pull_up:
                message = [PrivateConstants.PULLUP, pin, value, time_stamp] 
            else:
                message = [PrivateConstants.INPUT, pin, value, time_stamp]

            if last_value != value:
                if self.digital_pins[pin].cb:
                    await self.digital_pins[pin].cb(message)

            port_data >>= 1
```
I have not fully tested this but it seems to work. 
By the way, you will get a callback if the state of the pin is high since initially the state of the pin is set to a zero.

teucrus, 2020-04-15:
The pin type is insignificant.
Some more info for the multiple reporting:
This will not happen for pin 39 or pin 48 only for pins 40 to 47
I was thinking this has something to do with the pin groupping or ports for the arduino side?
I am very new with arduinos and don't know much about them. Hope this helps.

MrYsLab, 2020-04-15:
Please check my previous response. I think our responses may have crossed and you may not have seen my "fix".

teucrus, 2020-04-15:
> You can try changing the async def _digital_message(self, data): method to this:
>
> ```
>     async def _digital_message(self, data):
>         """
>         This is a private message handler [...]

Understood, I will test and report back.
Thank again.

teucrus, 2020-04-15:
Both fixes seem to be working now.
Thank you for your help.

MrYsLab, 2020-04-15:
Close with release v1.12

closed 2020-04-15 by MrYsLab