MrYsLab/pymata-express #17: Initializing the code without a connected arduino board.

teucrus, 2020-04-22:
Hello again.
I am at point of the developement where I need to test parts of my code without a physical device connected to it.
I would like to be able to instantiate the PymataExpress class without starting it.
I found the autostart=False option, but in the classes __init__, it will try to connect to the USB port.
I have moved that part out of __init__ and implemented the same mechanism as you used for the , autostart=False

I was wondering if this can become a standard part of the library, or if you have some other better ideas of what I need to do.

Thank you again for the great library and for you time.

MrYsLab, 2020-04-22:
If you are using any of the "standard" test runners to test your code:
[pytest-asyncio](https://github.com/pytest-dev/pytest-asyncio)
[asynctest](https://github.com/Martiusweb/asynctest/)
[alt-pytest-asyncio](https://github.com/delfick/alt-pytest-asyncio)

you may wish to create mocks for the test runner to stub out pymata-express.

If you are not using one of these test-runners currently, then the simplest thing would be to continue using your modified version of pymata-express. 

Changing the behavior of pymata-express may break existing code for some users, so I am reluctant to make the suggested changes.

I hope this helps.

MrYsLab, 2020-04-23:
I will be closing this issue, but if you wish to continue the discussion, you may post here.
Thanks.

closed 2020-04-23 by MrYsLab