MrYsLab/pymata-express #24: Robohat MM1 usage

mockbot, 2020-05-30:
while testing firmata express, I've noticed some issues with port mapping.

MM1 ports are defined here: https://robohatmm1-docs.readthedocs.io/en/latest/hardware/pinout/

e.g. using servos
#SERVO MM1 mapping to PIN
SERVO1=2
SERVO2=3
SERVO3=4
SERVO4=5
SERVO5=6

Who prepares the mapping of names and ports ?
Cian or you ?

wallarug, 2020-05-30:
Hi @MrYsLab ,

I checked this and the pin mappings appear correct in Arduino Board Definition.

How does FirmataExpress define the SERVO pins?  Do they start at D0?   This is not the same behaviour as Firmata which appear to start at D2.

MrYsLab, 2020-05-30:
pymata4/FirmataExpress does not assign a pin mode to any pins. Before using a pin the application assigns the mode using pymata4 calls and standard Arduino pin numbers to do so.

So to answer @mockbotw's question, the application, in this case, a Robohat MM1 Python API, would provide the mapping. But there are questions to be answered before designing that API. 

One approach would be to limit the user's view specifically and only to MM1 terminology. This may imply that the pins are configured in a preset way. Either hardcoded or through a config file. Or it may imply that the pin mode is not configured and the user does this through one of the set_pin_mode methods of pymata4, but instead of providing a pin number, an MM1 pin designation would be used.

I would think the latter would keep things as flexible as possible.

wallarug, 2020-05-30:
I know Arduino pre-sets some of the pin functions in the pinMap array (varient.cpp).

I would prefer option 2 if possible.  It's completely up what you think is best @MrYsLab

MrYsLab, 2020-05-30:
I agree flexibility is the way to go. This will need some investigation since I am not sure yet how Firmata views the pin mapping. Hopefully, the Firmata reports will answer my questions.

MrYsLab, 2020-05-31:
@mockbot Are you expecting the API to be asyncio, since this repository (pymata-express) is an asyncio API?
I prefer to get a "traditional" (non-asyncio) API accomplished first, and then port that work over to an asyncio version if required. I am also planning on building a [OneGPIO](https://mryslab.github.io/python_banyan/#gpio_intro/) version so that I can provide [Scratch3](https://mryslab.github.io/s3-extend/) interface.

mockbot, 2020-05-31:
Hi Alan,
for me it doesn't matter, if asyncio or traditional ;-) I have always
problems with different new boards and pin mappings.
Your new OneGPIO is very interesting and seems to solve these problems ...
Have you already received the MM1 hardware from Cian ? You will have the
same problem, like me.
Cian uses names for ports like SERVO1 or RCC1... now I have to find the
used pins by try end error, checking each available GPIO port for response.
This is no fun :-(
Pymate4 cannot help me to find the usage of pins :-(
Maybe OneGPIO has such function to get any capability and name from lower
level assignments.

BYI: I've build a small own prototype (PoC)  with simple JSON communication
between board and PC layer. The board sends own capabilities as collection
and PC software is able to use these names....

BUT, I have to check your OneGPIO concept first !!!!!
Please let me know, if I should test of verify something in the future.

Enjoy the power of MM1, I like this piece of flexible hardware very much

Cheers
Chrimo

mockbot, 2020-05-31:
Hi Alan,
I like your banyan concept very much.... it's so clear and well known
Like ODBC or Perl/DBI/DBD... unique API to the upper layer software and
drivers for individual proprietary boards.

Where is  your naming from ? Banyan, the old magic tree ? I saw a lot of
them on Hawaii and there was an old networking protocol in the 80th...
Banyan Vines :-)

Cheers
Chrimo

christian.mock.chrimo@gmail.com>:

mockbot, 2020-05-31:
Hi Alan,

while reading this: https://mryslab.github.io/python_banyan/#
I think, I read ROS1 documents :-)
https://wiki.ros.org/

FYI:
All the problems/pitfalls  of ROS1 are now eliminated and corrected in
ROS2, using DDS
https://design.ros2.org/articles/ros_on_dds.html

CU
Cheers
Chrimo

MrYsLab, 2020-05-31:
Hi Chrimo,
   I received the board, loaded FirmataExpress on it, got it to blink the board LED from a pymata4 script running in Linux. But then I plugged the board into a Raspberry Pi 4, and now can no longer see the serial port. I am working with Cian to see if I can reload the boot loader.
    Since you are much further along than I, from your description, it appears that the PA pin numbers on the Robohat are not logical pins but rather MCU physical pin numbers. Am I correct with that assumption?
    There is a report that may help to show how pins are defined, but the information is totally dependent upon the information contained in boards.h. If you go to the examples directory of either pymata4 or pymata-express and run retrieve_capability_report.py, a list of pins and the capability for each pin printed on the console. I would be curious to see if this maps to what you are seeing.

   The name of Banyan is somewhat of a stretch. Since python_banyan is built on top of ZeroMQ and an application is just a collection of (somewhat) independent python_banyan components or nodes, a banyan application reminded me of the banyan tree with many roots shooting up to make a single tree. 

I am aware of ROS  and I believe at one time they considered ZeroMQ but for some reason rejected it. I looked at the original version but it did not provide what I needed and was fairly complex. Most likely, I would have selected ROS2 had it been around when I started playing with ZeroMQ. 

Banyan has a long history of evolution. The [concept started out using JavaScript](https://github.com/MrYsLab/xi) and the JohnnyFive library. JohnnyFive in concept was fine, but at the time support for anything other than an Arduino was a  mixed bag. I then moved on to the [next incantation using Python.](https://github.com/MrYsLab/xideco). This was geared to building hardware interfaces for Scratch. FInally came banyan and then the concept of OneGPIO. 

My current thinking about the RoboHAT API is to provide a class that derives from pymata4 (or pymata-express). In this way, I can provide a view of the hardware that makes use of RoboHAT naming conventions to provide a hopefully intuitive API, but at the same time, allow the user to have access to the native API of pymata4. Looking at the GPS pins made me think this is probably the right way to go. FIrmata does not support a software serial interface for non-AVR processors. This means the API cannot support the GPS tx and rx pins of the RoboHAT. Having access to the full pymata4 API would allow one with an i2c GPS device write code using the standard pymata4 i2c API calls to access the device. I am going to prototype using this inheritance scheme in the next day or so just to see how viable it actually is.

mockbot, 2020-05-31:
Hello Alan,

thanks a lot for your feedback and detailed explanations - sounds great for
me !!!

GPS and NMEA sentences are a special case and option.....
Maybe it should/must  be handled like other special hardware DHT and
Ultrasonic Sensors....
but this is too much work for 2 users only :-)
What about implementing an internal virtual i2c device ? It's some kind of
proxy for special devices :-)
A simple read/write function for virtual internal dpseudo dummy devices can
solve this problem ... LOL.

Greetings and Thanks from Hamburg/Germany
Chrimo

wallarug, 2020-05-31:
Good Morning 😄 

> it appears that the PA pin numbers on the Robohat are not logical pins but rather MCU physical pin numbers. Am I correct with that assumption?

What are you referring with Logical Pins?   

The Robo HAT for Arduino is using the hardware pin numbers because that is what Arduino IDE required.   

If you look in the variants.cpp file, I think this is what you are referring to logical pins.  

https://github.com/robotics-masters/mm1-hat-arduino/blob/master/custom_board/RoboHat-0.0.26/variants/robohatmm1_m4/variant.cpp

> Looking at the GPS pins made me think this is probably the right way to go. FIrmata does not support a software serial interface for non-AVR processors.

I do not understand this comment.  The GPS Serial is hardware serial.

mockbot, 2020-05-31:
Question special feature GPS:

GPS is connected via serial UART and send NMEA strings

isn't it possible to use FIRMATA_SERAIL_FEATURE ? This allows access to
byte streams for second UART interface (often used at leonardo or mega
arduinos)
#ifdef FIRMATA_SERIAL_FEATURE
serialFeature.handlePinMode(pin, PIN_MODE_SERIAL);
#endif

MrYsLab, 2020-05-31:
What I mean by logical pins is Arduino's concept of having pin alias's. For example on an Uno, pin A0 is also pin 14. 

I was not aware of the variant file you provided the link to. After installing the Robohat into the Arduino IDE using the board's manager, that file is not in the arduino IDE directory tree. Should it be? Again, I am not that familiar with the Arduino low-level directory structure.

I was not aware that the GPS interface on the RobotHat used a UART and thought it was a bit-banger interface. My ignorance - I apologize. Currently, FirmataExpress only supports a single serial interface for the purposes of sending and receiving Firmata messages. Perhaps once everything else is working, I can explore getting the additional serial port working within FirrmataExpress.

wallarug, 2020-05-31:
@MrYsLab  I understand now.   That is based on the file I sent before.

Additionally, I started this last night to try and make things clearer.
https://robohatmm1-docs.readthedocs.io/en/latest/guide/Arduino%20API/Arduino_API/

If you are interested in learning more about the backend, I have this guide.
https://www.hackster.io/wallarug/arduino-ide-creating-custom-boards-89f7a6

The file you mentioned should be in the Arduino file structure.  Normally it is in %APPDATA%/local/Arduino15/ on a Windows Machine.

MrYsLab, 2020-06-01:
@wallarug Thanks for pointing out the path. Again, never having to deal with the low-level complexities of the Arduino IDE, I never knew that the IDE has yet a third directory. The files are stored in the .arduino15 directory.

MrYsLab, 2020-06-01:
@mockbot Supporting multiple serial ports is not a feature of StandardFirmata, but it looks like it has the ifdefs to turn this on. I will explore doing so for FirmataExpress, but the sketch is getting dangerously low on space for AVR boards, so I will have to see if adding the feature blows space capacity. If there is enough space in the Robohat but not in AVR boards, then I can conditionally compile for the Robohat.
I try to get to this within the next few days.

mockbot, 2020-06-01:
Hi Alan,

GPS is an option only, it's not builtin by default.

You would help me a lot, when you find something to use well known GPIO
ports (the mapping, e.g. RCC1->D0 ????) :-)
RCC1-4 ==> ???
SERVER01-8 ==> ???
BUTTON (Mainpower Switch) ===>???

Any chance to use names at pymata4 ?
Please check the attached Robohat MM1 PINOUT MAPS

Stupid question: is it possible to define portnames at board description
file e.g. Boards.h ?

Or is thee any equivalent at pymata4 level ??? #include Boards.py ?

RCC1=PINnumber
...
SERVO1=PINnumber
...
POWER_BUTTON=PINnumber
....

Thanks for all your investigations or some ideas to make endusers happy
Cheers
Chrimo

MrYsLab, 2020-06-01:
@mockbot wallarug posted [this link](https://github.com/robotics-masters/mm1-hat-arduino/blob/master/custom_board/RoboHat-0.0.26/variants/robohatmm1_m4/variant.cpp) that explains the pinouts very nicely. Please take a look and see if it answers your questions.

I had a chance to try compiling Firmata support for additional serial ports. It won't fit into the Uno and Leonardo, but it seems to be OK for the Mega2560. It compiles for the RoboHAT with no issue. Keep in mind, I have never used this feature in Firmata, so I have no idea if it actually works or not. I will need to add commands in pymata4 to support configuration and communication with those serial ports. If I can get it to work, the communication with pymata4 and an RPi should be able to be achieved without any additional cables. 

I am now rethinking my rethinking and am not sure if really need to provide a name mapping for the pins. If the user has a table with info similar to that in the link above, then they could set a pin with multiple personalities such as digital in, digital out and PWM, by pin number without me having to provide the various pin modes for a given pin. The user would just use the standard set of set_pin_mode API calls.

Were you able to run the retrieve_capability_report.py script and does it show a mapping similar to that in the link?

mockbot, 2020-06-02:
Hi Alan,

Yes, I digged in the file from the link....

BUT, it helped not much....

#SERVO MM1 mapping to PIN

SERVO1=2

SERVO2=3

SERVO3=4

SERVO4=5

SERVO5=6

...

I'm using now this source
https://github.com/robotics-masters/mm1-hat-arduino/blob/master/custom_board/RoboHat-0.0.26/variants/robohatmm1_m4/variant.h

for my mapping.

I'm preparing a pymata4 include, to use names...

But I'm still testing LOL...

This night, I've shut up MM1 while trying to use good old firmata_test UI
and firmata express with 57kB....

I had to restore the bootload file again :-(

Today, I will continue testing with names and pin numbers...

I think, we are on a very good way.

Thanks and greetings from Hamburg/Germany

Cheers

Chrimo

mockbot, 2020-06-02:
Hi Alan, Hi Cian,

here are the results of different capability reports: MM1, MICRO,NANO,UNO

MM1:

Retrieving Arduino Firmware ID...

Arduino Firmware ID: 1.1 FirmataExpress.ino

Retrieving analog map...

Auto-discovery complete. Found 46 Digital Pins and 7 Analog Pins

Capability Report

-----------------

Pin 0:

     Digital_Input: 1

     Digital_Input_Pullup: 1

     Digital_Output: 1

     Stepper: 1

     HC-SR04_Sonar: 1

     PWM: 8

     Servo: 14

Pin 1:

     Digital_Input: 1

     Digital_Input_Pullup: 1

     Digital_Output: 1

     Stepper: 1

     HC-SR04_Sonar: 1

     PWM: 8

     Servo: 14

Pin 2:

     Digital_Input: 1

     Digital_Input_Pullup: 1

     Digital_Output: 1

     Stepper: 1

     HC-SR04_Sonar: 1

     PWM: 8

     Servo: 14

Pin 3:

     Digital_Input: 1

[... 1464 lines cut ...]
     Analog_Input: 10

     Servo: 14

     I2C: 1

Pin 19:

     Digital_Input: 1

     Digital_Input_Pullup: 1

     Digital_Output: 1

     Stepper: 1

     HC-SR04_Sonar: 1

     Tone: 1

     Analog_Input: 10

     Servo: 14

     I2C: 1

wallarug, 2020-06-02:
Thanks @mockbot ,

Alan - if there is one thing I contribute to this project - it will be making that come out a lot nicer.   It's like 20 pages long!!

MrYsLab, 2020-06-02:
@wallarug @mockbot I have created a private repository and have invited both of you as contributors. I will be moving this issue to that repository so that we can continue to have conversations without confusing the current pymata-express users ;-).