MrYsLab/pymata-express #26: detect board hardware

mockbot, 2020-06-02:
Is there any chance to detect the board hardware, where pymata express  is running ?
e.g. Robot Hat MM1, Arduino UNO, Arduino NANO, Arduino Micro

I'm searching for an identifier to select automatic name to pin mapping :-)

MrYsLab, 2020-06-02:
There is nothing in Firmata that provides board identification, however, you may associate both the pymata-express (or pymata4) instance with an instance id that can be programmed into FirmataExpress. This is documented [here](https://mryslab.github.io/pymata-express/guidelines/#parameter-arduino_instance_id). This will guarantee that the program you are running will be associated with the correct com port. You would need to maintain the mapping of Arduino instance IDs and board types if you desire to do so.

Just to summarise, if you have multiple Arduinos plugged into your computer at the same time, by assigning a unique Arduino instance ID# in each FIrmataExpress instance (you will need to manually change the sketch before flashing), and then using the Arduino Instance ID  when instantiating the pymata-express class, the correct board should be found automatically.

This feature is something I created and is not part of StandardFirmata.

mockbot, 2020-06-02:
can ARDUINO_INSTANCE_ID be string ?
ARDUINO_INSTANCE_ID="MM1"
ARDUINO_INSTANCE_ID="NANO"
ARDUINO_INSTANCE_ID="UNO"

OR

integer

MrYsLab, 2020-06-02:
Actually it is considered by the [Arduino side to be a byte ](https://github.com/MrYsLab/FirmataExpress/blob/eff00dba368b10286a6bbfe753b0de44850e1362/examples/FirmataExpress/FirmataExpress.ino#L589), but handled as [an integer on the Python side.](https://github.com/MrYsLab/pymata-express/blob/1aafa864532a4bebbc2cd269f200b87f549f4099/pymata_express/pymata_express.py#L388)

closed 2020-06-02 by MrYsLab