MrYsLab/pymata-express #28: Analog pins cannot be set as digital input pullup.

teucrus, 2020-07-02:
Hello,

Please consider the following code:

```
import asyncio
from pymata_express.pymata_express import PymataExpress

async def setup(board):
	await board.set_pin_mode_digital_input_pullup(5,callback=my_callback)
	await board.set_pin_mode_digital_input_pullup(9,callback=my_callback)
	await board.set_pin_mode_digital_input_pullup(8,callback=my_callback)
	await board.set_pin_mode_digital_input_pullup(7,callback=my_callback)

async def my_callback(data):
	mode, pin_number,pin_state,timestamp=data
	print("-------------------------")
	print(mode,pin_number,pin_state,timestamp)
	print("-------------------------")

if __name__ == "__main__":
	board = PymataExpress()
	loop = asyncio.get_event_loop()
	loop.create_task(setup(board))
	loop.run_forever()

```

When I run this code and short the pins to ground nothing happens.
If I change the pins to actual digital pins, the code works as expected.
I know it's common to use the analog pins as digital but the arduino allows for it and in the current application this cannot be avoided.
If I run a similar code direcltly on arduino it works as expeced.

I have used the latest firmata-express firmware and pymata-express lib at the time of this writting.
Please adivse if a workaround can be done.

Thank you.

MrYsLab, 2020-07-02:
I believe this is not pymata-express issue, but a StandardFirmata issue. If you look at [this issue ](https://github.com/firmata/arduino/issues/432)against Arduino Firmata it may explain what you are seeing. Since FirmataExpress is derived from StandardFirmata it would behave the same way. Here is the explanation from that issue:

> Something that is hard to grasp is that Firmata does not have the equivalent of Arduino's analogRead() and digitalRead(), instead Firmata uses a reporting scheme where the user enables reporting of either an individual analog pin value or pin changes within a "digital port" (not to be confused with the microcontroller concept of a "port" -- in Firmata a "port" is a sequence of 8 digital pins in order starting from pin 0). If the user enables reporting for a port, any pin values in that port that have changes since the last iteration of the main loop are sent (that is far too frequent to report all digital pin values regardless of if they have changed). If the user enables reporting for an analog pin, that pin value is reported at the rate specified by the sampling interval (19ms by default), regardless if the pin value had changed.

teucrus, 2020-07-02:
I am sorry but I do not understand the connection betwwen my issue and that one.

I just want to set the A5-A10 as a  digital_input_pullup. and get the pin status only when it changes.
Would enable_analog_reporting help with this? Apologies I am a bit confused. it's late here.

Thank you.

MrYsLab, 2020-07-02:
There is no analog_pull up mode. When you are setting the input mode for A5-10 what are the pin numbers you are using? If you are using 5-10 that won't work, you need to provide the digital equivalent pin number. Using the Uno as an example, if you want to set A0 as a digital input with pullup enabled, the pin number is 14 and not 0.

teucrus, 2020-07-02:
How do I get the digital pin equivilant?
I have ATMEGA 2560 board.

I tried analog_read_map and got this:

`[127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15]`

teucrus, 2020-07-02:
Please close this. Found the solution. The pin# is +54 of the A#
Thank you for your efforts.

teucrus, 2020-07-03:
I used this code to calculate the correct pin number:

```
if "A" in pin_number:
	pin_number = int(pin_number.replace('A', ''))+54
else:
	pin_number = int(pin_number)
```

So when I want to use an analog pin, now I can use the A# and I don't get confused with conversions.
Since you read the board at the startup, it should be quite easy to change the offset and have the users put the actual analog input number when they want to use the input an a digital input.

MrYsLab, 2020-07-03:
Thanks for the suggestion. Something similar was considered early on, but the idea was rejected because:
1. There is a large installed base and adding this at this time may confuse users, or worse yet, break existing applications.
2. It adds complexity to the client - currently, all pins, both digital and analog are represented as integers. Using the "A" designation means not only having to add the parsing code when setting a pin but also when data is returned in a callback. What pin designation then is the correct one? Firmata does not return pin numbers as integers. It would mean that the client would then have to do a conversion.
3. I could allow both methods for analog pin designation, but then again, I need to keep track of the method the user picked when indicating the pin number in the callback.

So, I really do appreciate your suggestion, but for the reasons above I will not be implementing it. Using your solution at the application level is where I think it is properly handled.

Thanks again

closed 2020-07-03 by MrYsLab

teucrus, 2020-07-03:
Sorry to open the matter again. One last input on the matter.
You only need to apply the conversion one way when setting the pin as digital input or output (since we are interested in the digital equivalent). When the pin is reported you don't need to change the reported pin number. The conversion must be done is such a way that if someone inputs the actual digital equivalent it will also accept that. This way nothing will break.

Or at least in the set_pin_mode for digital add a note in the documentantion for this case. And maybe add somewhere how to find the digital equivalent of the analog inputs. Or even better add maybe an extra optional function to do that.

eg.: set_pin_mode_digital_input_pullup(analog_to_digital(1))

offtopic:: So I don't open a new issue. Could you please advise if it's possible to use pymata express for mfrc522 rfid scanner?
If so how can I aproach this?

Thank you.

MrYsLab, 2020-07-03:
Updating the documentation is a reasonable request and I will do that. Handling the conversion will be left to the application. I will try to add some guidance in the documentation as well.

I am not familiar with mfrc522, but it appears that it can be controlled either via SPI or i2c. Firmata does not support SPI. Since Firmata does provide i2c primitives, to support the device you would need to port a working i2c library to the Firmata primitives. There are no changes needed to either FirmataExpress or pymata-express.
Perhaps [this library](https://github.com/semaf/MFRC522_I2C_Library/blob/master/MFRC522_I2C.cpp) might be a good porting base.

If you would like to see an example of an i2c port, you can look [at this example.](https://github.com/semaf/MFRC522_I2C_Library/blob/master/MFRC522_I2C.cpp)

Quickly looking at this library, you would call the equivalent Firmata primitives for the following library methods:
 PCD_WriteRegister()
PCD_ReadRegister()
PCD_SetRegisterBitMask()

Some i2c devices are quite complicated and can be quite tricky to port such as this port for the[ MMA8452Q accelerometer.](https://github.com/MrYsLab/xibot/blob/master/redbot/redbot_accel.py). The MFRC522 appears to be of medium complexity, but you should judge that for yourself.

MrYsLab, 2020-07-03:
The documentation has been [updated.](https://mryslab.github.io/pymata-express/pin_modes/#mapping-analog-pin-numbers-to-digital-pin-numbers)