MrYsLab/pymata-express #30: SDIST at Pypi is missing at leat one file

onkelbeh, 2020-09-15:
Hi,

first, I hope you don't mind I included your component for [Home Assistant Gentoo Overlay](https://github.com/onkelbeh/HomeAssistantRepository).

I don't use your component myself, just noticed an error during regular compiliation tests, so I just wanted you to know that in Pypi's SDIST some files, at least one, are/is missing:

 ```
* Package:    dev-python/pymata-express-1.17
 * Repository: HomeAssistantRepository
 * Maintainer: b@edevau.net
 * Upstream:   MisterYsLab@gmail.com
 * USE:        abi_x86_64 amd64 elibc_glibc kernel_linux python_targets_python3_8 test userland_GNU
 * FEATURES:   network-sandbox preserve-libs sandbox userpriv usersandbox
>>> Unpacking source...
>>> Unpacking pymata-express-1.17.tar.gz to /var/tmp/portage/dev-python/pymata-express-1.17/work
>>> Source unpacked in /var/tmp/portage/dev-python/pymata-express-1.17/work
>>> Preparing source in /var/tmp/portage/dev-python/pymata-express-1.17/work/pymata-express-1.17 ...
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/dev-python/pymata-express-1.17/work/pymata-express-1.17 ...
>>> Source configured.
>>> Compiling source in /var/tmp/portage/dev-python/pymata-express-1.17/work/pymata-express-1.17 ...
 * python3_8: running distutils-r1_run_phase distutils-r1_python_compile
python3.8 setup.py build -j 10
Traceback (most recent call last):
  File "setup.py", line 3, in <module>
    with open('pypi_desc.md', "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: 'pypi_desc.md'
 * ERROR: dev-python/pymata-express-1.17::HomeAssistantRepository failed (compile phase):
 *   (no error message)
 *
```
The github archive compiles fine, for now [I changed the SRC_URI to Github](https://github.com/onkelbeh/HomeAssistantRepository/commit/2f9812c7542f17e0ee23fdbe1241a29b0f6dac66).

Perhpas this should be fixed.
Thanks a lot.

MrYsLab, 2020-09-15:
Thanks for letting me know. I did not know that sdist does not pickup all the .md files. I've added a MANIFEST.in file and bumped the release to 1.18. The file now appears in .gz.

closed 2020-09-15 by MrYsLab

DaAwesomeP, 2020-09-15:
@onkelbeh If you want to open an issue on the HA side, I can work on a pull.