MrYsLab/pymata-express #31: Digital pin reading not working

TheDatadrainer, 2020-09-17:
Hello.
I have some problem with a few function of this library.
By example, if I want to do something as trivial as getting the state of a button.
Initializing the pin like that with DPIN_BUTTON = 2 and arduino_button_pressed(data) as the callback function:
`await board.set_pin_mode_digital_input(DPIN_BUTTON, callback = arduino_button_pressed)`
Executing the code and pushing the button never execute the callback function.
Then if I try to read the pin directly with await board.digital_read(DPIN_BUTTON), the value is always 0.
And finally if I try to get the state of the button to know if it is configured as an INPUT with state = await board.get_pin_state(DPIN_BUTTON), the execution freezes on that instruction...
Maybe I'm doing something wrong, but I don't know what. So your help is more than welcome. Thank you.

ps : I'm using FirmataExpress 1.1.4 on an Arduino UNO and pymata-express 1.18 on Python 3.8.5

MrYsLab, 2020-09-17:
Have you looked at this [example](https://github.com/MrYsLab/pymata-express/blob/master/examples/digital_input.py)? Try changing the pin number and it should work for you. If you are still having issues, please post your code.

TheDatadrainer, 2020-09-17:
Thank you for your answer.
Yes, I tried the button on digital port 5 and changed the code consequently without success. What is strange is by using the non asynchronous version (pymata4) everything works great. And beside that the output works fine (I'm using tone and pwm_output)
I tried your example file and I have the same exact problem as I have with my code. Nothing happen on the digital_read function.

MrYsLab, 2020-09-17:
I found the issue and it is only in Windows. I am assuming that is what you are using. When Python 3.8 first came out, I had to add code since Python made changes to asyncio that broke things. It looks like they have now fixed things, which means my code is now broken. Until I am able to publish the fix, if you remove lines [111, 112](https://github.com/MrYsLab/pymata-express/blob/4bb2ae65f72f1f2b94caa86f77043e26660892c8/pymata_express/pymata_express.py#L111), and lines [266,267](https://github.com/MrYsLab/pymata-express/blob/4bb2ae65f72f1f2b94caa86f77043e26660892c8/pymata_express/pymata_express.py#L266) in pymata_express.py that will solve the issue. 

I should have a new release available in a day or so.

Thanks for letting me know about this.

MrYsLab, 2020-09-17:
This should now be fixed with release 1.19. I am closing the issue, but if you are still having problems you may either open a new issue, or post a comment here.
Again thanks for making me aware of this.

closed 2020-09-17 by MrYsLab