MrYsLab/pymata-express #32: Why it is so complicated to use pymata-express?

houdini52, 2020-10-26:
Hello,

do we have to define each function all the time to program something. I thought that all those functions were still programmed when we import pymata-express at the top of a program. !! ??? Can we use also the way that we program with python3 ?  Excuse me but I'm a newbee with python.  I tried to use pymata because I was not able to program to read a distance sensor (Ultrasonic). If I want to put a condition  (if) to brighten a led ex: on digital port 13 , how can I transfer the value of the data received (I mean the distance) if I want to activate something if the distance in cm is equal or shorter than 4 cm ?  A help should be very appreciate. Thank you in advance!
[distance_sensor.py.txt](https://github.com/MrYsLab/pymata-express/files/5436469/distance_sensor.py.txt)

MrYsLab, 2020-10-26:
Hi,
   Thanks for the question. Before getting to the answer, I just want to mention a few things. 
1. Since you are new to Python, you probably should be using the [pymata4](https://github.com/MrYsLab/pymata4) library instead of pymata-express. The documentation for pymata4 may be found [here](https://mryslab.github.io/pymata4/), and its API is [here](http://htmlpreview.github.io/?https://github.com/MrYsLab/pymata4/blob/master/html/pymata4/index.html). If you decide to use pymata4, you will not have to change the sketch on the Arduino, since both pymata4 and pymata-express use the same sketch.

The reason I am recommending pymata4 is that pymata-express uses an advanced feature of Python 3 called asyncio. The pymata4 library does not use asyncio and therefore is simpler to use.

2. I will be closing this issue, but you may still add comments here and I will see them. Also, please feel free to email directrly at the email address on this [page](https://github.com/MrYsLab) under my "picture".

Now on to your question. I think you may be confusing the actual library functions with those in the examples. The examples are just short programs that demonstrate using the API functions. To see the API functions for pymata4, just click on the link above.

Below is a pymata4 example to do what I think you wish to do. Please let me know if you have any questions about the program or the library. 

Thanks again for the question.

```
import sys
import time
from pymata4 import pymata4

"""
This program continuously monitors an HC-SR04 Ultrasonic Sensor
If the distance measured is less than or equal to 4 cm, then
turn on an LED.
"""

# index into callback data list to retrieve the sonar distance
DISTANCE_CM = 2

# Pin Definitions - change these to meet your needs
# HC-SRO4 Pins
TRIGGER_PIN = 8
ECHO_PIN = 9

# LED Pin
LED_PIN = 13

# General Definitions
ON = 1
OFF = 2

# A callback function to display the distance
def the_callback(data):
    """
    The callback function to display the change in distance
    :param data: [pin_type=12, trigger pin number, distance, timestamp]
    """
    # print(f'Distance in cm: {data[DISTANCE_CM]}')
    if data[DISTANCE_CM] <= 4:
        board.digital_write(LED_PIN, ON)
    else:
        board.digital_write(LED_PIN, OFF)

def demo(my_board, trigger_pin, echo_pin, led_pin, callback):
    """
    Set the pin mode for a sonar device. Results will appear via the
    callback.

    :param my_board: an pymata express instance
    :param trigger_pin: Arduino pin number
    :param echo_pin: Arduino pin number
    :param led_pin: Pin number of LED to be used for output
    :param callback: The callback function
    """

    # set the pin mode for the trigger and echo pins
    my_board.set_pin_mode_sonar(trigger_pin, echo_pin, callback)

    # set the pin mode for the LED
    my_board.set_pin_mode_digital_output(led_pin)

    # turn off LED
    my_board.digital_write(LED_PIN, OFF)

    # wait forever
    while True:
        try:
            time.sleep(.01)
        except KeyboardInterrupt:
            my_board.shutdown()
            sys.exit(0)

board = pymata4.Pymata4()
try:
    demo(board, TRIGGER_PIN, ECHO_PIN, LED_PIN, the_callback)
    board.shutdown()
except (KeyboardInterrupt, RuntimeError):
    board.shutdown()
    sys.exit(0)

```

closed 2020-10-26 by MrYsLab

houdini52, 2020-10-26:
Thanks a lot MrYsLab.  You know that it is not always interesting to answer to a newbee in Python! I want to thank you for answering me so rapidly.  I did what you suggest and install pymata4 instead of pymata-express.  I tested your program and is does exactly what I want.  Thank you for that, i learn a lot with examples.  I'm trying to make a SmartHOme prorotype to show youths aged 10 to 13 years old, in classrooms, the way that arduino and python can solve problems without using a RaspberryPy.  The challenge here is to make a loop where all sensors can work together to activate servo motors, lights (Leds), unlock doors with finger sensor or RFID sensor, detect the presence with Ultrasonic sensor as well as Infrared sensor.  I hope that my choice to make that with your library is the answer of my failures of the past 3 weeks. :-)  I hope to share my experience with other as soon as my program will be completed.  Thanks again to be present for me.

MrYsLab, 2020-10-26:
No problem. If you have any other questions, please don't hesitate to ask.