MrYsLab/pymata-express #33: Non yielding asyncio digital writes?

sheinbergon, 2021-04-19:
I just want to be sure I'm not missing something here.

I'm trying to drive two step motors directly from pymata-express, using a single arduino mega-pro, connected over a serial-usb port
I'm using the StandardFirmata to digital write step pulses directly, without step-motor extensions.

The step-motor driver is set to 400 steps per revolution. It takes me ~2000 ms to perform 2000 steps (5 revolutions, implying 1ms per step) when driving a single motor, each step consisting of 2 digital writes (LOW and HIGH). However driving two motors doubles the time it takes to complete the revolution, very much to my surprise.

_**The reason I'm surprised is that I expected digital writes should, given an 8 core core-i7 10875H CPU, quickly yield once the write was sent to the Arduino, meaning my the code should be able to allow the other digital writes (to other step motor) to at least partially execute, as local execution time are expected to be much faster than over-the-(serial)-wire communication**_  

So my question is - does the serial write to the Arduino itself does block/lock somewhere underneath the hood? I'm guessing there's a real constraint that's preventing concurrently accessing the same serial port, but I didn't  expect a 2X slowdown as a result of trying to drive 2 step motors. 

Sample code:

```python
import asyncio
import datetime
import logging

from pymata_express.pymata_express import PymataExpress

__MOTOR_1_POWER_PIN = 27
__MOTOR_1_DIRECTION_PIN = 29
__MOTOR_1_STEP_PIN = 31

__MOTOR_2_POWER_PIN = 17
__MOTOR_2_DIRECTION_PIN = 19
__MOTOR_2_STEP_PIN = 21

__LOW = 0
__HIGH = 1

__STEPS = 400 * 5
__ITERATIONS = 3

logging.basicConfig(level=logging.DEBUG)

async def setup_motor_pins(board: PymataExpress, power: int, direction: int, step: int) -> None:
  for pin in [power, direction, step]:
    await board.set_pin_mode_digital_output(pin)
  for pin in [direction, power]:
    await board.digital_pin_write(pin, __HIGH)

async def drive_motors(board: PymataExpress, step1: int, step2: int):
  for step in range(0, __STEPS):
    t1 = asyncio.create_task(motor_step(board, step1))
    t2 = asyncio.create_task(motor_step(board, step2))
    await asyncio.gather(t1, t2)

async def motor_step(board: PymataExpress, pin: int) -> None:
  await board.digital_pin_write(pin, __LOW)
  await board.digital_pin_write(pin, __HIGH)

async def main(board: PymataExpress) -> None:
  await setup_motor_pins(board, __MOTOR_1_POWER_PIN, __MOTOR_1_DIRECTION_PIN, __MOTOR_1_STEP_PIN)
  await setup_motor_pins(board, __MOTOR_2_POWER_PIN, __MOTOR_2_DIRECTION_PIN, __MOTOR_2_STEP_PIN)
  for iteration in range(0, __ITERATIONS):
    start = datetime.datetime.now().timestamp()
    await drive_motors(board, __MOTOR_1_STEP_PIN, __MOTOR_2_STEP_PIN)
    end = datetime.datetime.now().timestamp()
    print(f"{__STEPS} steps took {(end - start)} seconds to complete")

if __name__ == '__main__':
  board: PymataExpress = PymataExpress(com_port='/dev/ttyUSB1', baud_rate=57600)
  asyncio.run(main(board), debug=True)

```

MrYsLab, 2021-04-19:
Hi Idan,
    Pyserial does not provide asyncio functionality, so I wrote an [adaptation for serial write](https://github.com/MrYsLab/pymata-express/blob/4905c8ccaf88efa3802a2f1f8b330c45c9a38e07/pymata_express/pymata_express_serial.py#L68) that uses an asyncio _future_ that calls the pyserial write and then spins until the write is completed.

  I am not sure if this will help or not, but instead of using StandardFirmata, if you use [FirmataExpress](https://mryslab.github.io/pymata-express/firmata_express/), the link speed is double of that of StandardFirmata - it runs at 115200.

sheinbergon, 2021-04-19:
Hi @MrYsLab, thank you for the prompt response. 

This line seems completely synchronous https://github.com/MrYsLab/pymata-express/blob/4905c8ccaf88efa3802a2f1f8b330c45c9a38e07/pymata_express/pymata_express_serial.py#L81

Even if you loop asynchronously afterwards, the blocking op has taken place inside the event loop and outside of a future/task scope.

I'm guessing that that's where most of the time is spent

I can attempt to fix it if you'd be willing to receive a PR (running those blocking calls in a separate executor also feels like a viable option)

As for your second suggestion - it won't help with the current code. Things will always run slower if there's a blocking call inside one of the coroutines.

MrYsLab, 2021-04-19:
Please feel free to supply a pull request. If all looks good I will accept it, but your suggestion of using sockets might be easier and faster to implement. It requires using StandardFirmataWiFi on the Arduino.

sheinbergon, 2021-04-22:
@MrYsLab a small update

I've done some experimenting with replacing the use of pyserial with aioserial in your library. which should have taken care of code execution isolation. Due python's asyncio implementaion/nature it does **cost** more to asynchronously run this code in a separate executor as opposed to doing it serially, so I'm dropping this path. Optimistically sleep tunning when out_waiting is not empty also yields sub-optimal behavior, but that's due to python's sleep accuracy being bad once you go sub-milliseconds. that's mostly true with the synchronous `time.sleep` and gets even worse with `asyncio.sleep`..

In fact, `asyncio.sleep` can almost never accurately sleep less than a single ms, due to the internals overhead.

As for our use case, it's solved when using two arduino concurrently. We are able to achieve identical performance, from within a single pymata-express application. That will do for now.

As for you code, going over your code I noticed some sub-optimal behavior (send firmata/sysex message byte by byte over the serial port instead of sending them all together) as well as partial apis (sleep_tune not being  passed down the stack to the serial wrapper class). I can open a PR just for these concerns, should you like.

Let me know, 10x.

MrYsLab, 2021-04-22:
Thanks for taking the time to characterize the behavior. Python asyncio has a distance to go, but it has made great strides since it was introduced into an early version of Python3.

I am going to decline a PR at the moment since I am going to archive all the pymata clients in the near future and suggest that users implement their designs using [Telemetrix](https://mryslab.github.io/telemetrix/). Telemetrix is similar to Firmata, but does not use 7-bit bytes, and has a simpler implementation on both the client and server sides. BTW, Telemetrix sends the data in one string across the link as you suggest above for pymata. Performance should be similar to that of Pymata but provide a more consistent API.

I am closing this issue, but please feel free to add any additional comments, and I will see them.

Thanks again for all your efforts.

closed 2021-04-22 by MrYsLab

sheinbergon, 2021-04-22:
I'm understand your decision to decline a PR. As for archiving your clients - I have some questions regarding firmata and your implementation in specific I'd rather discuss in private. What's the best way to reach you?

MrYsLab, 2021-04-22:
You can find my email address under my avatar [here.](https://github.com/MrYsLab)