MrYsLab/pymata-express #41: Error compiling for board Arduino Uno.

tarlan-1907, 2022-05-07:


MrYsLab, 2022-05-07:
Please provide a screen dump of the error.

tarlan-1907, 2022-05-07:
[image]

MrYsLab, 2022-05-07:
I compiled version 1.2.0 using Arduino IDE 1.8.19 and am only getting a warning that can be ignored.
Make sure that you are using 1.2.0 and try upgrading to 1.8.19 of the IDE.

If you are still having issues, please copy and paste all of the errors/warnings shown in the bottom half of the IDE.

I am going to move this issue over to FirmataExpress
[image]
.